eembc/coremark #1: PORT_DIR error

ahmedawais, 2018-08-12:
make -j8 all 
/bin/sh: 1: [[: not found
/bin/sh: 1: [[: not found
Makefile:45: *** PLEASE define PORT_DIR! (e.g. make PORT_DIR=simple).  Stop.

i am using linux 32 bit and in make file its defined as PORT_DIR=Linux but still i am unable to buil in eclipse neon. please help

petertorelli, 2018-08-14:
You didn't specify a `PORT_DIR` so the `Makefile` attempted to infer one using `shell`, and assumes `bash`, e.g.:

~~~
# Ports for a couple of common self hosted platforms
UNAME=$(shell if [[ `uname 2> /dev/null` ]] ; then uname ; fi)
~~~

There are three solutions to your problem.

1. Manually specify `PORT_DIR` in the call to `make`, e.g.:

~~~
% make -j8 all PORT_DIR=linux
~~~

2. Set your default shell to `bash`

3. Add a line to the top of `Makefile` to change the default shell from `sh` to `bash`, e.g. `SHELL=/path/to/bash`

ahmedawais, 2018-08-16:
Hi,
 I have ESP32 device and using the Linux 32 bit, Eclipse IDE(neon). i have no experience with cross compilation setting up the flags and configuring the other parameter. 
ESP32 link: https://esp-idf.readthedocs.io/en/latest/api-reference/index.html

can you guide me ?