eembc/coremark #13: The use of DMULTITHREAD

ilya1725, 2019-06-24:
Hello:

What is the point of `-DMULTITHREAD`? I've tried running the benchmark on my system with this flag set to 1,2,4,16 and getting the same results.
However, the more complex underlined question is how CoreMarks takes into account multiple cores? There is nothing about it in the README or the EEMBC whitepaper.

Thank you.

benjaminogles, 2019-06-27:
You also have to specify which multithread implementation to use.
`-DUSE_PTHREAD=1` or `-DUSE_FORK=1` or `-DUSE_SOCKET=1` are the options I think.

ilya1725, 2019-06-27:
I used `-DUSE_FORK=1`. Same result.

petertorelli, 2019-06-27:
`-DMULTITHREAD=N` enables code for parallel execution, where the value of `N` determines the number of instances. The parallel implementation may be `USE_(FORK|PTHREAD|SOCKET)=1`. The implementation is defined in the `core_portme.c` file for the particular platform using the two functions `core_start_parallel` and `core_stop_parallel`. The `linux` and `linux64` targets both implement all three modes with example code. Here is a `pthreads` example: https://github.com/eembc/coremark/blob/2e9d6165f362f7071d29b7f88c6d32655e5d4a3c/linux64/core_portme.c#L236-L243

closed 2019-06-27 by petertorelli

rafi-kamal, 2024-07-09:
I am also having the same issue in Debian. I have to do `make clean` after every run, otherwise only a single thread gets used.