eembc/coremark #15: ITERATIONS variable in core_portme.mak not being obeyed

kenta2, 2019-09-19:
We can easily demonstrate this bug on linux64.  Modify linux64/core_portme.mak to add ITERATIONS=123456:

```
diff --git a/linux64/core_portme.mak b/linux64/core_portme.mak
index 5cfabee..4c7f3f5 100755
--- a/linux64/core_portme.mak
+++ b/linux64/core_portme.mak
@@ -138,3 +138,4 @@ MKDIR = mkdir -p
 # FLAG: PERL
 # Define perl executable to calculate the geomean if running separate.
 PERL=/usr/bin/perl
+ITERATIONS=123456
```

Then `make` to compile and run.  The log files show that (on my system) it ran 400000 iterations, not 123456 as requested.

The problem is the dollar parenthesis around ITERATIONS in the main Makefile, line 50.  After applying the patch from Pull Request https://github.com/eembc/coremark/pull/12 , the testcase works as expected, running 123456 iterations.

petertorelli, 2019-09-20:
Thanks for the test case. Didn't realize it was actually a bug and not just a simple typo.

petertorelli, 2019-09-20:
Merged in 6bb971c301390c86b6099e19bf4450307e735947

closed 2019-09-20 by petertorelli