eembc/coremark #17: coremark for rasberrypi

megha-an, 2019-11-08:
Hi

I am trying to run coremark benchmark on rasberry pi board , I ported the coremark to rasberrypi But when i run the code I am getting "segmentation fault" error . Board details are mentioned below

execution steps:

cd coremark
make linux
./coremark.exe

Board Details:
Linux pi64 4.19.66-v8-fc5826fb999e-p4-bis+ #2 SMP PREEMPT Fri Aug 16 13:58:31 GMT 2019 aarch64 GNU/Linux

Error :
"Segmentation fault"

I wanted to know what could be the reason .
If anyone know how to resolve the issue ,help me to resolve . 

Regards
Megha

petertorelli, 2019-11-08:
Which OS version uses aarch64v/8? (latest Buster10 is aarch7)

petertorelli, 2019-11-13:
Any update on this issue, @megha-an ? Where can I download the OS you used?

megha-an, 2019-11-15:
Hi 

I am using genpi64 os and you can download it in below link:

 https://github.com/sakaki-/gentoo-on-rpi-64bit

tommythorn, 2019-11-15:
Do you get the same issue under Ubuntu 18.04 LTS? (see https://www.google.com/amp/s/jamesachambers.com/raspberry-pi-ubuntu-server-18-04-2-installation-guide/%3famp)

petertorelli, 2019-11-15:
@megha-an 

You used the wrong `PORT_DIR`. Pointers are recast in the test, and the `linux` 32-bit version defines `ee_ptr_int` as a 32-bit type, then recasts as a pointer. This creates the segfault on 64-bit OSes, but also produces a warning which you overlooked (see below). Note I had to manually force `make PORT_DIR=linux` to recreate the error (had you not specified `PORT_DIR`, make would have deduced `linux64`).

E.g., if I force the `make PORT_DIR=linux`, I get this error:

~~~
demouser@pi64 ~/coremark $ ./coremark.exe 
ERROR! Please define ee_ptr_int to a type that holds a pointer!
Segmentation fault
~~~

Issuing `make` or `make PORT_DIR=linux64` works fine:

~~~
demouser@pi64 ~/coremark $ cat run1.log 
2K performance run parameters for coremark.
CoreMark Size    : 666
Total ticks      : 13453
Total time (secs): 13.453000
Iterations/Sec   : 8176.614881
Iterations       : 110000
Compiler version : GCC9.2.0
Compiler flags   : -O2 -DPERFORMANCE_RUN=1  -lrt
Memory location  : Please put data memory location here
			(e.g. code in flash, data on heap etc)
seedcrc          : 0xe9f5
[0]crclist       : 0xe714
[0]crcmatrix     : 0x1fd7
[0]crcstate      : 0x8e3a
[0]crcfinal      : 0x33ff
Correct operation validated. See README.md for run and reporting rules.
CoreMark 1.0 : 8176.614881 / GCC9.2.0 -O2 -DPERFORMANCE_RUN=1  -lrt / Heap
~~~

closed 2019-11-15 by petertorelli

megha-an, 2019-11-18:
Its working now ,
Thank you