eembc/coremark #20: Coremark use 2 fewer list items than we can

ScriptDevil, 2020-01-27:
In `core_list_init`, we use
```c
list_head *core_list_init(ee_u32 blksize, list_head *memblock, ee_s16 seed) {
	/* calculated pointers for the list */
	ee_u32 per_item=16+sizeof(struct list_data_s);
	ee_u32 size=(blksize/per_item)-2; /* to accomodate systems with 64b pointers, and make sure same code is executed, set max list elements */
	list_head *memblock_end=memblock+size;
	list_data *datablock=(list_data *)(memblock_end);
	list_data *datablock_end=datablock+size;
```
IIUC, per_item is calculated as 16B for list_head  + sizeof struct list_data_s.  size = capacity - 2 to provide space for head and tail; However, the value of size being used to compute `memblock_end` means we can only store capacity-2 elements even including head and tail; This also means the last 2 loop iterations of the code below would silently discard the data since we would be past `memblock_end` when inserting

```c
	/* create a fake items for the list head and tail */
	list->next=NULL;
	list->info=datablock;
	list->info->idx=0x0000;
	list->info->data16=(ee_s16)0x8080;
	memblock++;
	datablock++;
	info.idx=0x7fff;
	info.data16=(ee_s16)0xffff;
	core_list_insert_new(list,&info,&memblock,&datablock,memblock_end,datablock_end);
	
	/* then insert size items */
	for (i=0; i<size; i++) {
		ee_u16 datpat=((ee_u16)(seed^i) & 0xf);
		ee_u16 dat=(datpat<<3) | (i&0x7); /* alternate between algorithms */
		info.data16=(dat<<8) | dat;		/* fill the data with actual data and upper bits with rebuild value */
		core_list_insert_new(list,&info,&memblock,&datablock,memblock_end,datablock_end);
	}
```

This isn't a functional bug since `core_list_insert_new` is robust to the two drops and we don't use the pointer returned by that function in `list_init`.

I don't understand if this is actually intentional. However, I thought I should bring it up just in case my understanding was incorrect.

joseph-yiu, 2026-08-28:
I have checked with 2 people from Arm who were in EEMBC when CoreMark was developed, and they don't have any recollection of any discussion on this. So this behavior is likely to be accidental. Given that this is not a function bug and changing the behavior will affect results, the general view from Arm side is that we should leave it as is.

ruuddw, 2026-08-28:
Agreed, this just happens to be how it was defined back then so let's keep it.
Interesting finding, though. Something I expect an LLM to come up with, but this issue is from 2020...
It was classified as question back then as well, so no need to change anything.