eembc/coremark #25: undefined reference to pthread_*

jimsalterjrs, 2020-12-03:
When attempting to run all-threads on a Ryzen 7 3700 system:

```
user@box:~/coremark$ make XCFLAGS="-DMULTITHREAD=16 -DUSE_PTHREAD"
[...]
/usr/bin/ld: /tmp/cc1D8hXe.o: in function `core_start_parallel':
core_portme.c:(.text+0x199): undefined reference to `pthread_create'
/usr/bin/ld: /tmp/cc1D8hXe.o: in function `core_stop_parallel':
core_portme.c:(.text+0x1d0): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
```

jimsalterjrs, 2020-12-03:
-DUSE_FORK worked, for what that's worth; the problem seems limited to PTHREAD.

petertorelli, 2020-12-03:
The pthread library is not linked by default. I will add a note to the README. You need to modify the `core_portme.mak` file for your platform, adding `-lpthread` to `LFLAGS_END`. Example edit to the `linux64` folder:

~~~
:
#Flag: LFLAGS_END
#	Define any libraries needed for linking or other flags that should come at the end of the link line (e.g. linker scripts). 
#	Note: On certain platforms, the default clock_gettime implementation is supported but requires linking of librt.
LFLAGS_END += -lrt -lpthread
# Flag: PORT_SRCS
# Port specific source files can be added here
PORT_SRCS = $(PORT_DIR)/core_portme.c
:
~~~

closed 2020-12-03 by petertorelli

petertorelli, 2020-12-03:
@jimsalterjrs Also, thanks for pointing this out!

jimsalterjrs, 2020-12-03:
Still didn't get us there. 

```
gcc -O2 -Ilinux64 -I. -DFLAGS_STR=\""-O2 -DMULTITHREAD=16 -DUSE_PTHREADS -DPERFORMANCE_RUN=1  -lrt -lpthread"\" -DITERATIONS=0 -DMULTITHREAD=16 -DUSE_PTHREADS -DPERFORMANCE_RUN=1 core_list_join.c core_main.c core_matrix.c core_state.c core_util.c linux64/core_portme.c -o ./coremark.exe -lrt -lpthread
In file included from coremark.h:38,
                 from core_list_join.c:19:
linux64/core_portme.h:270:2: error: #error "Please implement multicore functionality in core_portme.c to use multiple contexts."
  270 | #error \
      |  ^~~~~
In file included from coremark.h:38,
                 from core_main.c:23:
linux64/core_portme.h:270:2: error: #error "Please implement multicore functionality in core_portme.c to use multiple contexts."
  270 | #error \
      |  ^~~~~
In file included from coremark.h:38,
                 from core_matrix.c:19:
linux64/core_portme.h:270:2: error: #error "Please implement multicore functionality in core_portme.c to use multiple contexts."
  270 | #error \
      |  ^~~~~
In file included from coremark.h:38,
                 from core_state.c:19:
linux64/core_portme.h:270:2: error: #error "Please implement multicore functionality in core_portme.c to use multiple contexts."
  270 | #error \
      |  ^~~~~
In file included from coremark.h:38,
                 from core_util.c:19:
linux64/core_portme.h:270:2: error: #error "Please implement multicore functionality in core_portme.c to use multiple contexts."
  270 | #error \
      |  ^~~~~
In file included from ./coremark.h:38,
                 from linux64/core_portme.c:21:
linux64/core_portme.h:270:2: error: #error "Please implement multicore functionality in core_portme.c to use multiple contexts."
  270 | #error \
      |  ^~~~~
linux64/core_portme.c:417:2: error: #error "Please implement multicore functionality in core_portme.c to use multiple contexts."
  417 | #error \
      |  ^~~~~
make[2]: *** [Makefile:86: compile] Error 1
```
This was with `core_portme.mak` edited as follows:

```
#Flag: LFLAGS_END
#       Define any libraries needed for linking or other flags that should come>
#       Note: On certain platforms, the default clock_gettime implementation is>
LFLAGS_END += -lrt
LFLAGS_END += -lpthread
# Flag: PORT_SRCS
# Port specific source files can be added here
PORT_SRCS = $(PORT_DIR)/core_portme.c

```

petertorelli, 2020-12-03:
The correct macro name is `-DUSE_PTHREAD`, not `-DUSE_PTHREAD**S**`.

jimsalterjrs, 2020-12-03:
Aha! Thank you in return for the typo catch. =)