eembc/coremark #34: Thread creation failure not tested

julienhascoet, 2021-06-18:
If thread create fails, it fails silently:

https://github.com/eembc/coremark/blob/1541482bf3e6ef7f5c69f5be76b14537b60833d0/core_main.c#L275
https://github.com/eembc/coremark/blob/main/posix/core_portme.c#L280

Error code should be checked.

petertorelli, 2021-07-01:
This will not be fixed for the following reasons:

1. The CRC is checked for each context. If a thread/fork does not succeed, then the CRC will be incorrect and the test will report an error. This is checked only if a known seed was used, and to generate an official CoreMark score, a known seed must be used.

2. EEMBC has a policy of not altering the code inside the timing loop, as it fundamentally alters what is being measured and invalidates all previous scores for a new version. Now, realistically, the amount of code executed to start a thread and check its return value should be insignificant and amortized over a significant number of loops (since the minimum runtime is 10 seconds). However, had the failure not been caught in the above caveat, it would be a more compelling argument to alter the inter-loop code.

Thanks for filing the issue. It will be queued for a minor-version release in the future.

julienhascoet, 2021-07-02:
Great, thanks a lot !