eembc/coremark #35: ee_u32 accessed with %d format specifier

mysterymath, 2021-10-20:
In core_main.c, the results of time_in_secs is printed using the %d format specifier, which handles arguments of type int.
However type secs_ret is type ee_u32, which is not int-sized on 16-bit platforms (for which I'm working on an LLVM port :) ).
I got lucky in that my platform is little endian, so the low order 16-bits that will be read by %d are the ones that I want, and there aren't any arguments after the problematic %d's.

Three ways off the top of my head to fix this:

1. Cast to int before the format specifier.
2. Use the C99 format specifier macros for uint32_t.
3. Create an EEMBC-specific version of the above macro and add it to the port headers.

mysterymath, 2021-10-20:
https://github.com/llvm-mos/llvm-mos

LLVM backend for the MOS 6502!

It's still really early days with respect to optimization work (so please don't pass this around too much ;) ), but I'm happy to report that a 1MHz MOS 6502 achieves 0.0386 CoreMarks/s with our backend.

Looking at the generated output, there's a ton of obvious inoptimalities. It's nice to have such a comprehensive benchmark to work against! Porting it was surprisingly easy, too.

joseph-yiu, 2026-10-17:
Closing this issue as we are finalizing new release. Thanks all for the contribution.

closed 2026-10-17 by joseph-yiu