eembc/coremark #40: coremark was impacted by some extra print

duchao713, 2022-03-31:
I added some print in the "main" function (after the "iterate" function and the total_time calculation, for sure), then I found that the coremark changed.

Is that reasonable ?
I thought it should not impact the coremark since I did not change the iterate process at all.

Thanks.

petertorelli, 2022-04-01:
Your inclination is correct, adding code outside the timing loop should not impact the score. If you could share your modifications it would help debug.

duchao713, 2022-04-06:
For example, I made a test on my machine (X86, Linux), with the latest coremark source code on 'main' branch.

I typed 'make' directly.

before any modifications, 'run1.log' shows:

`2K performance run parameters for coremark.
CoreMark Size    : 666
Total ticks      : 13759
Total time (secs): 13.759000
Iterations/Sec   : 29071.880224
Iterations       : 400000
Compiler version : GCC7.5.0
Compiler flags   : -O2 -DPERFORMANCE_RUN=1  -lrt
Memory location  : Please put data memory location here
			(e.g. code in flash, data on heap etc)
seedcrc          : 0xe9f5
[0]crclist       : 0xe714
[0]crcmatrix     : 0x1fd7
[0]crcstate      : 0x8e3a
[0]crcfinal      : 0x65c5
Correct operation validated. See README.md for run and reporting rules.
CoreMark 1.0 : 29071.880224 / GCC7.5.0 -O2 -DPERFORMANCE_RUN=1  -lrt / Heap`

then I deleted some prints (line 359-378, core_main.c)，then 'run1.log' shows:
`2K performance run parameters for coremark.
CoreMark Size    : 666
Iterations       : 400000
Compiler version : GCC7.5.0
Compiler flags   : -O2 -DPERFORMANCE_RUN=1  -lrt
Memory location  : Please put data memory location here
			(e.g. code in flash, data on heap etc)
seedcrc          : 0xe9f5
[0]crclist       : 0xe714
[0]crcmatrix     : 0x1fd7
[0]crcstate      : 0x8e3a
[0]crcfinal      : 0x65c5
Correct operation validated. See README.md for run and reporting rules.
CoreMark 1.0 : 29014.942695 / GCC7.5.0 -O2 -DPERFORMANCE_RUN=1  -lrt / Heap`

The total_time was already got before the print, so I thought it should not change the score.

duchao713, 2022-04-06:
It is only the fluctuation which is acceptable ?

closed 2022-04-06 by petertorelli