eembc/coremark #48: How to deploy known_id=2 in MCU's coremark test?

Cpantherk, 2022-11-14:
Dear,

I want to test the score using Coremark's test code for an MCU with RAM = 2KB. 
How should I deploy it?
How to calculate the coremark score for the test results?
Currently my test results are as follows.

CoreMark Size    : 170
Total ticks      : 15673
Total time (secs):
Iterations/Sec   :
Iterations       : 5000
Compiler version : GCC10.3.1
Compiler flags   : -Os
Memory location  : STACK
seedcrc          : 0xd967
[0]crclist       : 0x11ee
[0]crcmatrix     : 0x66cc
[0]crcstate      : 0x5cd9
[0]crcfinal      : 0x0a03
Cannot validate operation for these seed values, please compare with results on a known platform.

Thanks for your help in advance.
BR,
Cpantherk

petertorelli, 2022-11-14:
By default it is set up for 2K of memory (666 bytes per task * 3 tasks). The score is iterations/sec.

closed 2022-11-15 by petertorelli

Cpantherk, 2022-11-16:
Dear Petertorelli，

Thanks for your help.

BR，
Cpantherk