eembc/coremark #56: Library-function test-driver

Artoria2e5, 2023-10-12:
The current test driver is based on string (usually stdout and argv) I/O and is not quite convenient to call from more restrictive environments such as iOS. A more "`struct`-ured" API taking a struct input (multithread, seed) and a struct output (anything printed) might be more adequate and portable. `core_results` is 90% there already.

Artoria2e5, 2023-10-14:
The *less* intrusive way is to redirect `main` and `ee_printf`. Basically the caller calls the renamed `core_main` supplying argc and argv, then retrieve the result from a global buffer that the new printf writes to.

This can be achieved without modifying the 5 core files -- ifdef is enough to rename main(). The makefile still needs to be taken down and rewritten.

joseph-yiu, 2026-08-21:
@Artoria2e5 
A pull request has been created https://github.com/eembc/coremark/pull/81 to allow main() to be renamed.

joseph-yiu, 2026-10-17:
https://github.com/eembc/coremark/pull/81 is merged into development branch (https://github.com/eembc/coremark/blob/dev_2026q3/).
Closing this as we are finalizing new release. Thanks for the feedback.

closed 2026-10-17 by joseph-yiu