eembc/coremark #59: Error while running the coremark with gcc

WaseemEng, 2024-05-17:
When i run this command in terminal to run the coremark:
gcc -O2 -o coremark.exe core_list_join.c core_main.c core_matrix.c core_state.c core_util.c simple/core_portme.c -DPERFORMANCE_RUN=1 -DITERATIONS=1000
Please let us know in which directory we have to run this command that you have mention in Readme file.
[image]

sergiocaballeroo, 2024-08-08:
you solved the error? same issue in Windows Server
[image]

waseem-10xe, 2024-08-09:
Can you please add the path of these files where these are present so this error will be removed.

closed 2024-08-09 by waseem-10xe