eembc/coremark #65: Position Independent Coremark executable fails when not loaded at its static link time address

fontamsoc, 2024-11-08:
Position Independent Coremark executable fails as follow when not loaded at its static link time address:

```
2K performance run parameters for coremark.
[0]ERROR! list crc 0xbd52 - should be 0xe714
[0]ERROR! state crc 0x6f31 - should be 0x8e3a
CoreMark Size    : 666
Total ticks      : 1287594042
Total time (secs): 12.875940
Iterations/Sec   : 155.328460
Iterations       : 2000
Compiler version : GCC13.2.0
Compiler flags   : -O3 -DPERFORMANCE_RUN=1
Memory location  : STATIC
seedcrc          : 0xe9f5
[0]crclist       : 0xbd52
[0]crcmatrix     : 0x1fd7
[0]crcstate      : 0x6f31
[0]crcfinal      : 0xfbbb
Errors detected
```

I am curious to know what in the algorithm depends on Coremark getting loaded at its static link time address.
I am able to reproduce this same failure on two different cpu isa architecture.

oniot6, 2024-12-11:
```
static ee_u8 intpat[4][5]
    = { "5012", "1234", "-874", "+122" };
static ee_u8 floatpat[4][9] = { "35.54400",
                              ".1234500",
                              "-110.700",
                              "+0.64400" };
static ee_u8 scipat[4][9]   = { "5.500e+3",
                            "-.123e-2",
                            "-87e+832",
                            "+0.6e-12" };
static ee_u8 errpat[4][9]   = { "T0.3e-1F",
                            "-T.T++Tq",
                            "1T3.4e4z",
                            "34.0e-T^" };
```

fontamsoc, 2024-12-14:
That explains it; thanks !

closed 2024-12-14 by fontamsoc